gbionics/adam #7: Numpy error when install ADAM via pip on 20.04 with python3-numpy apt package installed

traversaro, 2021-11-30:
I am installing ADAM via the installation process, and the following error is printed:
~~~
ERROR: jaxlib 0.1.73 has requirement numpy>=1.18, but you'll have numpy 1.17.4 which is incompatible.
ERROR: jax 0.2.25 has requirement numpy>=1.18, but you'll have numpy 1.17.4 which is incompatible.
~~~

It seems that pip finds the `numpy` installed via apt and so does not installs it in the virtualenv, but then it complains that the version is not correct.

traversaro, 2021-11-30:
While testing the examples, all of them fail with:
~~~
ModuleNotFoundError: No module named 'numpy.typing'
~~~

So I guess that indeed there is a dependency on a specific numpy version?

Giulero, 2021-12-06:
Hi @traversaro!
 
> So I guess that indeed there is a dependency on a specific numpy version?

Indeed, it seems! 

Also, which version of python are you using?

traversaro, 2021-12-06:
> Also, which version of python are you using?

~~~
traversaro@IITICUBLAP257:~$ python3
Python 3.8.10 (default, Sep 28 2021, 16:10:42)
[GCC 9.3.0] on linux
~~~

Giulero, 2021-12-13:
@traversaro sorry for the late response! Do you think that bumping the NumPy version to > 1.18 in the [`setup.cfg`](https://github.com/ami-iit/ADAM/blob/main/setup.cfg) would solve the problem?

traversaro, 2021-12-13:
> Do you think that bumping the NumPy version to > 1.18 in the [`setup.cfg`](https://github.com/ami-iit/ADAM/blob/main/setup.cfg) would solve the problem?

I do not know. I tried to apply:
~~~
(adamenv) traversaro@IITICUBLAP257:~/ADAM$ git diff
diff --git a/setup.cfg b/setup.cfg
index cc135cd..cedd2c6 100644
--- a/setup.cfg
+++ b/setup.cfg
@@ -9,7 +9,7 @@ license_file = LICENSE
 packages = find:
 python_requires = >=3.7
 install_requires =
-        numpy
+        numpy >= 1.18
         scipy
         casadi
         matplotlib
~~~

and installing with `pip install .` in a clean environment indeed installs numpy 1.21, but if I then stash the modifications and install again, numpy 1.21 is again installed, so I am not sure what is happening here (I am not a big pip expert to be honest).

PR #14 (2021-12-17): Fix numpy typing dep
I've bumped the NumPy version to 1.20 since [NumPy Typing](https://numpy.org/devdocs/reference/typing.html) requires NumPy >= 1.20. 

I'm using this PR also to extend a bit the `setup.cfg` and to remove 2 useless prints spotted in the tests.

Fixes #7
closed 2021-12-17 by Giulero