gbionics/adam #15: ADAM Christmas/Holiday Release?

traversaro, 2021-12-23:
Hello @Giulero @gabrielenava, 
  given that ADAM is getting quite a few users (great!) do you think it could make sense to do a tag/release (even just something like `v0.0.1`) to simplify [scientific reproducibility](https://the-turing-way.netlify.app/reproducible-research/overview.html)? Note that even if you tag a release, as long as you document that the public interface can change, you can always change the public interface, so no need to worry like "let's wait for the software to be ready for doing a release". I mention this as tipically software, especially research software, is never ready.

traversaro, 2021-12-23:
fyi @DanielePucci

Giulero, 2021-12-23:
HI @traversaro! 
It would be nice! I would like to have nice documentation (with [sphinx](https://www.sphinx-doc.org/en/master/)?) and some additional examples, but they can be part of a new release.
Do we need further actions (apart from documenting non-stable API)?

traversaro, 2021-12-23:
>  but they can be part of a new release.

Exactly! Notice that if you want to keep it low key for now, to not give the impression to users that reach the repo that there is a release ready to be used, you can also just tag the repo without creating a GitHub release. 

>  Do we need further actions (apart from documenting non-stable API)?

I was looking if there was someway to specify the version in the Python configuration files, but I could not find any. At the moment when you pip install it get installed as `0.0.0`, i.e. the output of `pip list | grep ADAM`  is:
~~~
ADAM              0.0.0
~~~ 

Perhaps there is some place in which we can set that the version is now 0.0.1 ?

traversaro, 2021-12-23:
I did a quick check, and adding a `version` line in the `setup.cfg`  seems to work correctly:
~~~diff
diff --git a/setup.cfg b/setup.cfg
index 83f25bb..95f9d07 100644
--- a/setup.cfg
+++ b/setup.cfg
@@ -5,6 +5,7 @@ author = "Giuseppe L'Erario"
 author_email = gl.giuseppelerario@gmail.com
 license_file = LICENSE
 url = https://github.com/ami-iit/ADAM
+version = 0.0.1

 keywords =
     robotics
~~~

Not sure if this is the best/most idiomatic way, but for sure it works.

Giulero, 2021-12-23:
Thanks @traversaro! 
Opening a PR.

Giulero, 2021-12-23:
ADAM's first release is [here](https://github.com/ami-iit/ADAM/releases/tag/v0.0.1)!

DanielePucci, 2021-12-23:
/evviva

traversaro, 2021-12-23:
Great!

closed 2021-12-23 by traversaro