gbionics/adam #26: Package in conda-forge

traversaro, 2022-08-02:
For packaging in projects that install all their deps via conda-forge without resorting to PyPI, I would like to package this via conda-forge . This should be trivial to do, generating the recipe via grayskull ( https://github.com/conda-incubator/grayskull ). Unfortunatly the tricky aspect is that there is no [urdf-parser-py](https://github.com/ros/urdf_parser_py) package on conda-forge, so we should first package that dep in conda-forge.

Giulero, 2022-08-26:
Hi @traversaro! I missed this, sorry! Indeed it would be nice to ship the library via conda-forge :)
I could start packaging `urdf-parser-py` (maybe I'll need some advice from you :D).

traversaro, 2022-09-12:
We hitted this with @DanielePucci today, even if the work around of just using pip to install inside the conda environment worked fine.

traversaro, 2023-01-29:
I proposed the recipe in conda-forge in https://github.com/conda-forge/staged-recipes/pull/21894 .

traversaro, 2023-05-26:
This was fixed:
* https://github.com/conda-forge/staged-recipes/pull/21894
* https://github.com/ami-iit/ADAM/pull/43

> We hitted this with @DanielePucci today, even if the work around of just using pip to install inside the conda environment worked fine.

I do not remember anything related to this, but basically now you can install it in an existing environment:
~~~
mamba install -c conda-forge adam-robotics
~~~
or create a new environment with:
~~~
mamba create -n adamenv -c conda-forge adam-robotics
~~~

closed 2023-05-26 by traversaro