gbionics/adam #28: Support prismatic joints

traversaro, 2022-10-03:
For the goals of one of our internal projects, it would be convenient if ADAM supported prismatic joints. 

cc @Giulero

DanielePucci, 2022-10-03:
CC @CarlottaSartore

Giulero, 2022-10-03:
It shouldn't be difficult. I'll do it!

CarlottaSartore, 2022-10-03:
In the past, I worked on it (see for instance https://github.com/ami-iit/ADAM/compare/main...CarlottaSartore:ADAM:add_joint_type_param), I can try to make a minimum invasive PR only with the modification for supporting the prismatic joint. 

For the time being, I would disregard the hardware parameters since it brings with it the problem of the modality in which the joint origin should be moved  (see https://github.com/ami-iit/element_hardware-intelligence/issues/72)

traversaro, 2022-10-03:
Thanks @Giulero @CarlottaSartore ! I think @VenusPasandi was also interested in working on this, so probably you can coordinate.

Giulero, 2022-10-03:
Created a PR for this. I used the code @CarlottaSartore  implemented in a previous PR <3

I also might need an urdf of a robot with prismatic joints.

PR #29 (2022-10-05): Add prismatic joints
This PR adds prismatic joints manipulations. Basically I copy-pasted the modification @CarlottaSartore did in a previous pr.

Since we now support 3 different types of joint it might be worth to generalize the joint interface, since the code contains some duplicates. I'll do it (maybe in a separate PR).

I need a urdf of a robot with prismatic joints to test the modifications.

Fixes #28
closed 2022-10-05 by Giulero