gbionics/adam #32: Broken Action for ADAM Main

CarlottaSartore, 2022-12-21:
In the current master, the actions are broken see https://github.com/CarlottaSartore/ADAM/actions/runs/3693902157/jobs/6366061917

I think the cause could be in `numpy` which version `1.24.1` has just been release (see https://pypi.org/project/numpy/#history)

I am trying to fix the versioning in b60087e58dcf9da42af58a60826d37624bf62222 but still encountering some errors 

C.C. @Giulero  @traversaro

traversaro, 2022-12-21:
For the future: especially if we install packages from a rolling release distro such as PyPI or conda-forge without any pin, it could make sense to also have a scheduled job to run the CI every n days, so that we catch regressions like this, see for example:
* https://github.com/robotology/idyntree/blob/v8.0.0/.github/workflows/ci.yml#L6
* https://github.com/robotology/robotology-superbuild/blob/v2022.11.0/.github/workflows/ci.yml#L14

traversaro, 2022-12-21:
> I am trying to fix the versioning in [b60087e](https://github.com/ami-iit/ADAM/commit/b60087e58dcf9da42af58a60826d37624bf62222) but still encountering some errors

For reference, the latest successful build used numpy 1.23.*, at least it seems:
* https://github.com/ami-iit/ADAM/actions/runs/3190152767/jobs/5204871876
* https://github.com/ami-iit/ADAM/actions/runs/3712303882

CarlottaSartore, 2022-12-21:
Yes  I have opened PR  #33 (to separate from the hardware parameters implementation), that should fix and it works with numpy 1.23.5

PR #34 (2022-12-22): Fix numpy dep
Fixes #32
closed 2022-12-22 by Giulero