gbionics/adam #50: Jax installation for GPU

stergiosba, 2023-11-08:
```pip install adam-robotics[jax]```

traversaro, 2023-11-08:
fyi @Giulero @flferretti

flferretti, 2023-11-09:
By default, the installation of JAX comes with the CPU version of `jaxlib`. If you're interested in the GPU version, I'd suggest to install it as per the [JAX documentation](https://jax.readthedocs.io/en/latest/installation.html#pip-installation-gpu-cuda-installed-via-pip-easier):

```python
pip install --upgrade "jax[cuda12_pip]" -f https://storage.googleapis.com/jax-releases/jax_cuda_releases.html ```

stergiosba, 2023-11-09:
Personally I am aware of the installation process for using the GPU with Jax. I just wanted to point this out so that others are not confused when their code is not as fast as they hoped.

traversaro, 2023-11-09:
> Personally I am aware of the installation process for using the GPU with Jax. I just wanted to point this out so that others are not confused when their code is not as fast as they hoped.

Good point, would you be interested (if it is ok for the mantainer @Giulero) to add a sentence in the readme warning about this and perhaps linking to https://github.com/google/jax#instructions ?

Giulero, 2023-11-09:
Thanks @stergiosba for opening the issue! @flferretti is right.
Yes @traversaro, I agree!

Giulero, 2024-06-19:
I guess this is solved. Closing!

closed 2024-06-19 by Giulero

traversaro, 2024-06-19:
As an additional info for anyone finding this via Google, since ~May 2025 the `jax` package on conda-forge on Linux amd64 by default installs a working GPU-enabled jax, i.e.:

~~~
conda create -n adamjax -c conda-forge adam-robotics jax
~~~

on a `linux-64` system with a working NVIDIA driver installation install a GPU-power jax out of the box.