gbionics/adam #54: Rename this repo to adam-robotics?

traversaro, 2023-12-06:
As the package is installed as `adam-robotics` in PyPI and conda-forge, perhaps for discoverability and clarity it is better to rename this repo to `adam-robotics` as well?

@Giulero

traversaro, 2023-12-11:
Any opinion on this @Giulero ?

Giulero, 2023-12-11:
Hi @traversaro! Sorry for the late reply. You're right! However, honestly, I do not like the name `adam-robotics` a lot, but this is only a personal preference!  
I was forced to use this name in PyPI since obviously the name `adam` was already taken :smile: 
I can start to add a tag but if there is a general consensus we can also change the name!

traversaro, 2023-12-11:
Ok, as you prefer, for discoverability it is easier to use the same name, but if you do not like it no problem, it was only a suggestion. What do you mean with "I can start to add a tag"?

Giulero, 2023-12-11:
> What do you mean with "I can start to add a tag"?

Wrong naming! I mean `topic`, the tags we add in the repo description.

traversaro, 2023-12-11:
Ahh, I do not think it help a lot, but for sure better then anything. I think we can close this issue then.

Giulero, 2023-12-11:
> Ahh, I do not think it help a lot

Me neither, honestly :smiling_face_with_tear:

traversaro, 2023-12-15:
Another suggestion: can we at least rename it `adam` instead of `ADAM` ? At least we are consistent between `import adam` and the name of the repo.

Giulero, 2023-12-15:
Yeah, it's a nice idea! I also have to change some instances in the readme.

traversaro, 2023-12-15:
> Yeah, it's a nice idea! I also have to change some instances in the readme.

Ok, can I proceed with the rename ADAM/adam?

traversaro, 2023-12-15:
Ok,  I see th repo was renamed, great.

PR #61 (2023-12-15): Update README.md changing the capitalization (ADAM -> adam)
This PR homologates the repo name to the package one, changing the name from **ADAM** to **adam**.
Fixes #54.
closed 2023-12-15 by Giulero