gbionics/adam #58: Differentiability w.r.t what exactly?

JonathanKuelz, 2023-12-12:
Dear ADAM developers,

I just stumbled across your package and it looks amazing! However, there's a question I couldn't find an answer to in your examples and the README: If I am not mistaken, a JAX/PyTorch-implementation of the algorithms you included in ADAM should allow to compute gradients not only with respect to the joint configuration (as far as I understand, that's what you do in your examples), but leverage autograd to do so w.r.t any model parameter (e.g. joint offsets (aka link length) or inertial parameters).

Is that something that is supported by ADAM and if so, would you be so kind to provide me a pointer on where to start with something like that?

traversaro, 2023-12-12:
cc @Giulero @CarlottaSartore

Giulero, 2023-12-13:
Hi @JonathanKuelz! Thanks for your interest and for opening the issue :)
You're right! This feature has been just introduced by @CarlottaSartore in #49. 
We introduced a new module, `adam.parametric.casadi/pytorch/etcetc`, that computes rigid body algorithms using parametrized body information. 
You can start to have a look at the tests, by checking the `parametric` tests folder to see how the methods are called. 
Also, I'd warn that the `pytorch/jax` interfaces are not really stressed: if you have some doubts or spot bugs don't hesitate to come to us! 

Btw, @CarlottaSartore will tell you more about the rationale behind the `parametric` module!