gbionics/adam #73: Casadi,

tommasoandina1, 2024-03-25:
Hi, I'm calculating the robot's dynamic equations, but I'm encountering some difficulties in computing the torque forces, which represent the torques exerted by the robot's motor joints. Can someone please assist me?

traversaro, 2024-03-25:
Sure, feel free to ask any question you like. Which kind of difficulty are you experiencing in particular?

tommasoandina1, 2024-03-25:
I want to calculate the dynamics of my robot from this formula
Mddq + C + h = tau
where:
M  is the robot's mass matrix, representing the system's inertia and dependent on the joint positions q
ddq is the vector of joint accelerations.
C  is the matrix of Coriolis and centrifugal forces, dependent on the joint positions q and velocities dq
g  is the vector of gravitational forces, dependent on the joint positions q
tau is the vector of joint torques, representing the forces applied to the joints to control the robot.

My question is how can I calculate the tau?

Giulero, 2024-03-26:
Hi @tommasoandina1!

You can retrieve the quantities you calling the functions `mass_matrix_fun()`, `coriolis_term_fun()`, `gravity_term_fun()`, or if you want the coriolis term and the gravity term together, `bias_force_fun()`. 

Here a snippet for these computations
```python
import adam
from adam.casadi import KinDynComputations
import icub_models
import numpy as np
import casadi as cs

# fill the model_path with the string pointing to your urdf
model_path = "where your urdf is`
# The joint list you want to control/handle.
joints_name_list = [joint1_name, joint2_name, joint3_name]

kinDyn = KinDynComputations(model_path, joints_name_list, root_link)

# The base pose as 4x4 homogenous matrix. If you're using a fixed base manipulator, put the identity.
H = cs.DM.eye(4)
# The joint values
s = cs.SX.sym('s', num_dof)
# The base velocity, zero if you are using a manipulator
v_b = cs.DM.zeros(6) # or cs.SX.sym('v_b', 6)
# The joints velocity
s_dot = cs.SX.sym('s_dot', num_dof)
# The base acceleration, zero if you are using a manipulator
v_b_dot = cs.DM.zeros(6) # or cs.SX.sym('v_b_dot', 6)
# The joints acceleration
s_ddot = cs.SX.sym('s_ddot', num_dof)

# initialize to get the functions you need
mass_matrix_fun = kinDyn.mass_matrix_fun()
coriolis_term_fun = kinDyn.coriolis_term_fun()
gravity_term_fun =  kinDyn.gravity_term_fun()
# or if you want one term for C and G, bias_force_fun() = kinDyn.bias_force_fun()

M = mass_matrix_fun(H, s)
C = coriolis_term_fun(H, s, v_b, s_dot)
G = gravity_term_fun(H, s)

# robot dynamics
tau = M * cs.vertcat(v_b_dot, s_ddot) + C + G 

# if you want to wrap it in a cs.Function
tau_fun = cs.Function('tau_f', [H, s, v_b, s_dot, v_b_dot, s_ddot], [tau])
```

If anything is not clear do not hesitate to come back here!

closed 2024-03-26 by tommasoandina1