gbionics/adam #76: Use of the Adam

tommasoandina1, 2024-05-02:
hi I wanted to ask if I was making correct use of your library because I wrote the following repository
https://github.com/tommasoandina1/Doosan_h2515?tab=readme-ov-file

traversaro, 2024-05-02:
The code in https://github.com/tommasoandina1/Doosan_h2515/blob/main/doosan_h25215.py seems to make sense. Do you have any specific question on some points?

tommasoandina1, 2024-05-02:
Yep, I have to calculate the direct dynamics of the robot using this formula:
ddq = M^-1(tau-h), 
where h is bias force, M the Mass Matrix and tau the coppie al giunto calcolate dalla legge di controllo
Where the controll law is PD:
where h is the bias force, Mis the Mass Matrix, and tau are the joint torques calculated from the control law. The control law used is PD:
tau = Kp * (q_des - q) - Kd * dq,

where q_des is the desired joint position, q is the actual position, dq is the actual joint velocity, and Kp and Kd
Kd are gains that you need to set.
Once Kp is set, it's common practice to set  Kd = 2 * sqrt(Kp).

I have to choose Kp through a trial-and-error process, starting with very low values (e.g., 0.1) and increasing it until the system becomes unstable. In these simulations, you can try setting q_des to different values and verify that the controller can approach q_des closely enough (within 1 degree) in reasonable times (2/3 seconds).

Then, my next goal is to evaluate the simulation accuracy by varying the simulator's time step dt

	q_next = q + dt * dq
	dq_next = dq + dt * ddq

by computing the ideal behavior of the simulator using a very low dt value, like 1/16 ms. Then, increase the dt value by multiplying it by 2 each time and check how much the robot's trajectory (q, dq) deviates. Continue this process until reaching a range of dt up to 2 ms. I expect that above 1 ms, the simulator's accuracy will drastically worsen due to vibrations at 200 Hz. Please note that even if the simulator's time step is very low (0.1 ms), the frequency with which I need to update the tau value (i.e., the controller frequency) will be higher. For example, we could reasonably assume that the controller's period is 2 ms because I will later need to add forces that disturb my system.

closed 2024-05-06 by tommasoandina1