gbionics/adam #79: Test ADAM with conda-forge dependencies failing on macos-latest

Giulero, 2024-05-10:
The action running the conda-forge dependency test is failing when testing macos-lates: 
https://github.com/ami-iit/adam/actions/runs/9026258191/job/24803267789

traversaro, 2024-05-10:
The problem is probably that `macos-latest` now points to `macos-14`, that is `osx-arm64` (i.e. arm processors) instead of `osx-64` (intel processors). And apparently we do not have `urdfdom-py` on `osx-arm64` ?

traversaro, 2024-05-10:
This will be fixed by https://github.com/conda-forge/urdfdom-py-feedstock/pull/6 .

Giulero, 2024-05-10:
Thanks @traversaro !

traversaro, 2024-05-26:
> This will be fixed by [conda-forge/urdfdom-py-feedstock#6](https://github.com/conda-forge/urdfdom-py-feedstock/pull/6) .

This was merged, so the job should be green again the next time it runs.

Giulero, 2024-05-27:
Thanks a lot @traversaro for solving this ! :)

traversaro, 2024-05-27:
Green: https://github.com/ami-iit/adam/actions/runs/9248193736 .

closed 2024-05-27 by traversaro