gbionics/adam #90: Forward kinematics is so slow

richardrl, 2024-06-23:
Hi,
Is there any way to get a fast forward kinematics from this library?

Here's the speed for inverse kinematics and the speed for forward kinematics:

traversaro, 2024-06-24:
What is your use case? If you are not interested in batching, I guess `numpy` or `casadi` probably should give you a faster fk (for `casadi` especially if you are using common subexpression elimination, see https://github.com/ami-iit/adam/pull/86).

traversaro, 2024-06-24:
By the way, how are you computing `self.fast_c` in your snippet? To get some a fast optimization-based IK, I guess you have some sort of fast FK to use inside the cost function?

richardrl, 2024-06-24:
@traversaro Thanks for your comment - the problem was not using the jitted version of forward kinematics. It is close to 0 compute time with the jitted version.

closed 2024-06-24 by richardrl