gbionics/adam #98: Consider generalizing all backends of array APIs

diegoferigo, 2024-09-17:
I remember a f2f discussion a while ago with @Giulero about how to generalize algorithms to work on NumPy / JAX / PyTorch etc. At that time, there was no concrete solution excluding what today became [`Farama-Foundation/Jumpy`](https://github.com/Farama-Foundation/Jumpy) that was only supporting NumPy and JAX.

After a couple of years, it seems that the community is trying to address this use case with the [Python array API standard](https://data-apis.org/array-api/latest/). It seems to be that this solution is the one that got major traction and attracted most of the interest towards this unification.

ADAM, today, generalizes the underlying array APIs used by the RBDAs by introducing a custom abstraction layer based on `adam.core.spatial_math.{ArrayLike|ArrayLikeFactory|SpatialMath}`, which has implementations in CasADi, JAX, NumPy, and PyTorch.

I think that CasADi --at least in the short term and at least officially-- won't be compatible with these Array APIs. However, providing in ADAM a new backend based on these new APIs could ease the maintenance of the other backends (or, possibly remove them), and could bring with no additional effort also support for all [other array providers](https://data-apis.org/array-api/latest/purpose_and_scope.html#stakeholders) that are adhere with the API standard.

A good starting point could be the [`data-apis/array-api-compat`](https://github.com/data-apis/array-api-compat) wrapper[^1], its usage seems pretty straightforward.

[^1]: https://data-apis.org/array-api-compat/

Giulero, 2024-09-18:
Thanks a lot @diegoferigo! <3 
This would be lovely indeed. I agree that finally exploiting these new Array APIs will make the library more maintainable. 
Regarding CasADi, we could maybe maintain the actual abstraction layer and implement a new backend for what we can unify. A sort of hybrid structure, eventually converging to a cleaner solution when we find a solution for the CasADi backend. 
As soon as possible I can try to draft a PR and ask you other suggestions ;)

traversaro, 2025-07-25:
In a related news, SciPy just merged array API support for its Rotation module: https://github.com/scipy/scipy/pull/23249 .