gbionics/adam #115: Tests failing on conda-forge:macos-latest

Giulero, 2025-01-02:
The action testing adam with conda-forge dependencies fails when the test work is on `macos-latest`.

I'm not sure it depends on an update introduced in the last PR #111 since 
- the action was not failing on a scheduled run 3 days ago: https://github.com/ami-iit/adam/actions/runs/12554531648 
- the action then failed 2 days ago on another scheduled run:  https://github.com/ami-iit/adam/actions/runs/12566302910

traversaro, 2025-01-02:
For googlability, the error is:

~~~
 ==================================== ERRORS ====================================
___________ ERROR collecting tests/parametric/test_jax_parametric.py ___________
ImportError while importing test module '/Users/runner/work/adam/adam/tests/parametric/test_jax_parametric.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
../../../micromamba/envs/adamdev/lib/python3.13/importlib/__init__.py:88: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/parametric/test_jax_parametric.py:3: in <module>
    from adam.parametric.jax import KinDynComputationsParametric
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/adam/parametric/jax/__init__.py:5: in <module>
    from .computations_parametric import KinDynComputationsParametric
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/adam/parametric/jax/computations_parametric.py:7: in <module>
    import jax.numpy as jnp
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/jax/__init__.py:25: in <module>
    from jax._src.cloud_tpu_init import cloud_tpu_init as _cloud_tpu_init
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/jax/_src/cloud_tpu_init.py:17: in <module>
    from jax._src import config
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/jax/_src/config.py:27: in <module>
    from jax._src import lib
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/jax/_src/lib/__init__.py:87: in <module>
    import jaxlib.xla_client as xla_client
../../../micromamba/envs/adamdev/lib/python3.13/site-packages/jaxlib/xla_client.py:33: in <module>
    from . import xla_extension as _xla
E   ImportError: dlopen(/Users/runner/micromamba/envs/adamdev/lib/python3.13/site-packages/jaxlib/xla_extension.so, 0x0002): Symbol not found: __ZN4absl12lts_2024072212log_internal10LogMessagelsIPKvLi0EEERS2_RKT_
E     Referenced from: <05E679BA-8DDA-37BF-BA62-E2CC01CA7001> /Users/runner/micromamba/envs/adamdev/lib/python3.13/site-packages/jaxlib/xla_extension.so
E     Expected in:     <D623F952-8116-35EC-859D-F7F8D5DD7699> /Users/runner/micromamba/envs/adamdev/lib/libabsl_log_internal_message.2407.0.0.dylib
~~~

traversaro, 2025-01-02:
By the way, I recently saw another error related to jaxlib failing https://github.com/conda-forge/jax-feedstock/issues/167, not sure if it could be related.

traversaro, 2025-01-02:
The diff between working and not working environment is quite clear, considering also the nature of the error:

~~~diff
--- <unnamed>
+++ <unnamed>
@@ -32,7 +32,7 @@
   jax2torch           0.0.7       pyhd8ed1ab_0                  conda-forge
   jaxlib              0.4.35      cpu_py313hd85e7f4_0           conda-forge
   jinja2              3.1.5       pyhd8ed1ab_0                  conda-forge
-  libabseil           20240722.0  cxx17_hf9b8971_1              conda-forge
+  libabseil           20240722.0  cxx17_h07bc746_2              conda-forge
   libblas             3.9.0       26_osxarm64_openblas          conda-forge
   libboost            1.86.0      hc9fb7c5_3                    conda-forge
   libcblas            3.9.0       26_osxarm64_openblas          conda-forge
@@ -52,7 +52,7 @@
   libpng              1.6.44      hc14010f_0                    conda-forge
   libprotobuf         5.28.2      h8f0b736_0                    conda-forge
   libqdldl            0.1.7       hb7217d7_0                    conda-forge
-  libre2-11           2024.07.02  h2348fd5_1                    conda-forge
+  libre2-11           2024.07.02  h07bc746_2                    conda-forge
   libscotch           7.0.5       hc2b2845_4                    conda-forge
   libsqlite           3.47.2      h3f77e49_0                    conda-forge
   libtorch            2.5.1       cpu_generic_h4c131e9_7        conda-forge
@@ -93,7 +93,7 @@
   python_abi          3.13        5_cp313                       conda-forge
   pytorch             2.5.1       cpu_generic_py313_h44dfc17_7  conda-forge
   pyyaml              6.0.2       py313h20a7fcf_1               conda-forge
-  re2                 2024.07.02  hcd0e937_1                    conda-forge
+  re2                 2024.07.02  h6589ca4_2                    conda-forge
   readline            8.2         h92ec313_1                    conda-forge
   requests            2.32.3      pyhd8ed1ab_1                  conda-forge
   scipy               1.14.1      py313hc010ede_2               conda-forge
~~~

probably something broke ABI compatibility in the libabseil `cxx17_h07bc746_2` build.

traversaro, 2025-01-02:
The issue seems https://github.com/conda-forge/abseil-cpp-feedstock/issues/87 . I saw it a few days ago, but I did not realized it could have impacted us.

Giulero, 2025-01-03:
Thanks a lot @traversaro for the help! 
So if I understand correctly, there is nothing we can do adam-side.

traversaro, 2025-01-03:
> So if I understand correctly, there is nothing we can do adam-side.

On the adam side we can simply pin libabseil to a working version, see https://github.com/conda-forge/abseil-cpp-feedstock/issues/87#issuecomment-2568380408 . I can prepare a PR.

Giulero, 2025-01-03:
Right! Thanks a lot! :)

traversaro, 2025-01-03:
There is a candidate for a global conda-forge fix in https://github.com/conda-forge/abseil-cpp-feedstock/pull/88/files , at this point I think we can wait for it.

Giulero, 2025-01-03:
Cool! I agree ;)

traversaro, 2025-01-04:
I restarted the latest failing job in https://github.com/ami-iit/adam/actions/runs/12606669284 .

traversaro, 2025-01-05:
The issue is now fixed: https://github.com/ami-iit/adam/actions/runs/12615772775/job/35156215446 . 

It was fixed by https://github.com/conda-forge/abseil-cpp-feedstock/pull/88 and https://github.com/conda-forge/abseil-cpp-feedstock/pull/90 .

Giulero, 2025-01-07:
Super! Thanks a lot @traversaro!
Closing the issue then.

closed 2025-01-07 by Giulero