gbionics/adam #123: Request about adam and modifying urdf

PietroFerrario, 2025-01-20:
Hello, 
I'm working on a co-design project using adam to load the urdf file of a robot model. I'd like to ask if it possible to modify such a file, once it's loaded, directly using this library. Or if there is no such a feature and the only way to do it is to first modify the urdf and then to load it with adam.

CarlottaSartore, 2025-01-22:
Hello @PietroFerrario,

Thank you for opening an issue and for your interest in using Adam for co-design projects!

Regarding your question about modifying a URDF file after it has been loaded using Adam: while the library itself does not currently support direct modification of the URDF once it's loaded, there is an alternative approach that might fit your needs.

Adam provides a parametric implementation that allows you to parameterize the robot model based on specific hardware parameters, such as link lengths and densities. You can find more information about this functionality here:
[Parametric Implementation](https://github.com/ami-iit/adam/tree/main/src/adam/parametric)

This implementation enables you to derive dynamic and kinematic quantities parameterized by these hardware properties, effectively supporting customization of the robot model for co-design purposes. Once you have used Adam to identify the desired link hardware parameters, you can create a new URDF that reflects these changes using the [urdf-modifiers](https://github.com/icub-tech-iit/urdf-modifiers).

Additionally, you can explore detailed examples in the ADAM  test folder to understand how this parameterization works in practice:[Test Example Parametric](https://github.com/ami-iit/adam/blob/main/tests/parametric/)

We hope this answers your question. Let us know if this answer your question or if you need any additional information !

PietroFerrario, 2025-01-22:
Thank you @CarlottaSartore for your answer!