gbionics/adam #129: Incompatibility with Python 3.8: Use of Python 3.9+ type annotations (list[...], tuple[...]) leads to TypeError

johnny-ning, 2025-07-11:
Dear adam developers,

Thank you for your great work on the adam project!
I am using adam together with legged_gym and followed the installation instructions that specify using Python 3.8.

However, when running the code, I encountered the following error:

`TypeError: 'type' object is not subscriptable`

This happened at this line:

`def get_links(self) -> list[Link]:`

It seems that recent versions of the codebase are using Python 3.9+ type annotations like list[...], tuple[...], which are not supported in Python 3.8. In Python 3.8, these should be replaced with the corresponding typing types, for example:

```
from typing import List
def get_links(self) -> List[Link]:
```
Could you please clarify:

1.Is Python 3.8 still officially supported?

2.If so, would it be possible to revert to Python 3.8-compatible type annotations (typing.List, typing.Tuple, etc.)?

3.Or should we upgrade to Python 3.9+ despite the current documentation?

Thank you very much for your help!

Giulero, 2025-07-11:
Hi @johnny-ning! Thanks for opening the issue :) 
Yeah, as @traversaro mentioned, the documentation is outdated, and Python 3.8 is not supported anymore. I'll open a pr for the update!

johnny-ning, 2025-07-12:
Hi @traversaro and @Giulero, thanks for your quick reply!

Since Python 3.8 is no longer supported in the latest version of adam, I'm wondering:

Is there a specific older release or commit of adam that is known to work with Python 3.8 (e.g., before list[...] and tuple[...] syntax was introduced)?

I would like to use adam together with Isaac Gym, which currently only supports Python 3.8. If there's a suitable past version (maybe a specific tag or commit), I’d be very grateful if you could point it out.

Thanks again for your support!

Giulero, 2025-07-15:
Hi @johnny-ning! I see!

[v0.3.2](https://github.com/ami-iit/adam/releases/tag/v0.3.2) should be compatible with Python 3.8 (v0.3.2 should be the release where we introduced the "new" type annotations). Could you double check? 
Thanks a lot!

closed 2026-05-05 by Giulero